hideakitai/ArduinoOSC #1: No ESP8266 support?

himijendrix24, 2018-04-08:
Using a nodeMCU v2.

If I try to compile the example files, I get the error 
`#error UNSUPPORTED PLATFORM`

After adding
`#define ESP_PLATFORM 1`

it's still not working: 
`Poti_AP_OSC:27: error: 'osc' does not name a type
 osc.begin(udp, recv_port);`

hideakitai, 2018-04-08:
Sorry, I will remove error about `ESP8266`.
But maybe you can build it by changing:
- add `#define ESP_PLATFORM`
- rewrite WIFi related stuff based on your platform manner, or remove them

I could build it in ESP8266 Generic Module and NodeMCU v1.0
(I couldn't find v2 in board manager)

hideakitai, 2018-04-11:
@himijendrix24 I have tested on NodeMCU v1, and worked fine. Please try version 0.1.2. Thanks.

closed 2018-04-11 by hideakitai