hideakitai/ArduinoOSC #4: Implement Wildcards for addresses

CameronDevine, 2018-12-31:
Per the OSC 1.0 spec `*`, `?`, and strings such as `[a-c]` are used to match multiple OSC addresses. Are there any plans to support this section of the OSC spec? I am currently working on a project where it would be useful to do the following, 

```c
osc.addCallback("/ch/*/fader", &callback)
```

Currently I have to add a separate callback for each of the 16 channels on my mixer.

hideakitai, 2018-12-31:
Hi @CameronDevine , that's cool idea but currently I have no plan to support wildcards. Because Arduino does not support regex (> C++11). If you want to use wildcards in ArduinoOSC, you can use other regex libraries in Pattern::patternComp(OSCMessage& m).  For example, [slre](https://github.com/cesanta/slre) and [Regexp](https://github.com/nickgammon/Regexp).
I may integrate them in the future, if I would have completed to improve core functions... Thank you for your feedback!

closed 2018-12-31 by hideakitai

hideakitai, 2018-12-31:
By the way, I'm now planning to introduce [OSCPKT](http://gruntthepeon.free.fr/oscpkt/html/index.html) to handle osc messages in these days, so wildcards may come earlier than expected (I found OSCPKT supports wildcards now :))