hideakitai/ArduinoOSC #5: Compilation error when including ArduinoOSC.h in several headers

Zarfab, 2019-01-30:
I have a class that can handle OscMessage so I need to include ArduinoOSC.h in the header of this class. This leads to compilation error (multiple definition of Packetizer::crc8 and ArduinoOSC::match).
To solve this problem, I created two files (ArduinoOSC.cpp next to ArduinoOSC.h and Packetizer.cpp next to Packetizer.h), included their respective headers and moved the code from the problematic functions in the corresponding cpp.
It now compiles and works well even though I don't fully understand why.
Here are the concerned files.
[ArduinoOSC_modif.zip](https://github.com/hideakitai/ArduinoOSC/files/2811531/ArduinoOSC_modif.zip)

Thanks for this great and useful work by the way.

hideakitai, 2019-02-04:
@Zarfab Thank you for the feedback! I will check and fix it later.

hideakitai, 2019-03-31:
I'm so sorry for late reply, but I couldn't reproduce this problem, so I will close this issue this time. If you still have problem, please feel free to tell me and please give me a set of source codes. Thanks!

closed 2019-03-31 by hideakitai

cscashby, 2019-09-07:
I see the same issue when I have a .h file which includes ArduinoOSC.h, and is referenced in more than one source file.

I have solved it by putting the include to ArduinoOSC.h in my class implementation file, but this isn't ideal as it means I can't have OSC member variables in my class definition.

I can mock up a quick sample if needed.

hideakitai, 2020-06-02:
@cscashby Sorry for late reply! This issue is solved in latest version. Please try it!