hideakitai/ArduinoOSC #6: Core Panic

devinroth, 2019-03-22:
Whenever I send osc to my ESP32 using Arduino OSC it crashes. I get the following error. 

lambda : 192.168.0.169 61084 1 /lambda 3 Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4014298c  PS      : 0x00060830  A0      : 0x800d1eda  A1      : 0x3ffb1ee0  
A2      : 0x00000003  A3      : 0x00000001  A4      : 0x3ffbbaa8  A5      : 0x00000001  
A6      : 0x3ffba72b  A7      : 0x3ffafedb  A8      : 0x00000004  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3f4000ba  A12     : 0x0000000a  A13     : 0x3ffafedb  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000003  LBEG    : 0x4014298c  LEND    : 0x4014299a  LCOUNT  : 0x00000003  

Backtrace: 0x4014298c:0x3ffb1ee0 0x400d1ed7:0x3ffb1f10 0x400d3381:0x3ffb1f40 0x400d33b6:0x3ffb1f90 0x400d4b11:0x3ffb1fb0 0x4008e6f1:0x3ffb1fd0

hideakitai, 2019-03-30:
Could you give me more information?
- source code
- esp32-arduino version
- Arduino IDE or PlatformIO and version
- ArduinoOSC version
- etc.

devinroth, 2019-04-02:
Source: ArduinoOSC OscWIFi example program.
Esp32-arduino version: 1.0.1
Arduino IDE 1.8.9
ArduinoOSC version: 0.2.1

The board sends OSC messages just fine. It's only when receiving that it crashes. It seems to be related to the callback. It receives data just fine if I don't use a callback.

hideakitai, 2019-04-02:
mmm... in the same configuration, my examples works fine w/ Max8. 
Let me know more information:
- Is arduino-esp32 repo in the latest master branch??
- Which program do you use to send OSC?
Thank you for your cooperation!

hideakitai, 2019-04-02:
Let me add :
- Please give a source code. (Delete your WiFi info plz ;))

devinroth, 2019-04-02:
First off, thanks for helping. I'm pretty sure this problem is an ESP32 issue not a ArduinoOSC problem but let's try to get to the bottom of the problem.

Maybe there's a problem with the board. I have another ESP32 I can try it on later today.

I installed the arduino-esp32 with the board manager in the IDE as recommended. https://github.com/espressif/arduino-esp32

I'm using my own program [SwiftOSC](https://github.com/devinroth/SwiftOSC) to send data. I even tried sending identical data to the board as you send out in the example code with the same problem. So I know for a fact it's not a formatting issue. 

I'm using your OscWifi example with only changing Wifi info. I tried commenting out all the code in the callback so it happens somewhere between receiving the data and calling the callback. 

```
#include <ArduinoOSC.h>

// WiFi stuff
const char* ssid = "your-ssid";
const char* pwd = "your-password";
const IPAddress ip(192, 168, 1, 201);
const IPAddress gateway(192, 168, 1, 1);
const IPAddress subnet(255, 255, 255, 0);

// for ArduinoOSC
OscWiFi osc;
const char* host = "192.168.1.200";
const int recv_port = 10000;
const int send_port = 12000;

void onOscReceived(OscMessage& m)
{
    Serial.print("callback : ");
    Serial.print(m.ip()); Serial.print(" ");
    Serial.print(m.port()); Serial.print(" ");
    Serial.print(m.size()); Serial.print(" ");
    Serial.print(m.address()); Serial.print(" ");
    Serial.print(m.arg<int>(0)); Serial.print(" ");
    Serial.print(m.arg<float>(1)); Serial.print(" ");
    Serial.print(m.arg<String>(2)); Serial.println();
}

void setup()
{
    Serial.begin(115200);

    // WiFi stuff
    WiFi.begin(ssid, pwd);
    WiFi.config(ip, gateway, subnet);
    while (WiFi.status() != WL_CONNECTED) { Serial.print("."); delay(500); }
    Serial.print("WiFi connected, IP = "); Serial.println(WiFi.localIP());

    // ArduinoOSC
    osc.begin(recv_port);

    // TODO: TBD
    // osc.subscribe("/int32", i);
    // osc.subscribe("/float", f);
    // osc.subscribe("/string", s);
    // osc.subscribe("/blob", b);

    osc.subscribe("/callback", onOscReceived); // old style (v0.1.x)

    osc.subscribe("/lambda", [](OscMessage& m)
    {
        Serial.print("lambda : ");
        Serial.print(m.ip()); Serial.print(" ");
        Serial.print(m.port()); Serial.print(" ");
        Serial.print(m.size()); Serial.print(" ");
        Serial.print(m.address()); Serial.print(" ");
        Serial.print(m.arg<int>(0)); Serial.print(" ");
        Serial.print(m.arg<float>(1)); Serial.print(" ");
        Serial.print(m.arg<String>(2)); Serial.println();
    });
    osc.subscribe("/wildcard/*/test", [](OscMessage& m)
    {
        Serial.print("wildcard : ");
        Serial.print(m.ip()); Serial.print(" ");
        Serial.print(m.port()); Serial.print(" ");
        Serial.print(m.size()); Serial.print(" ");
        Serial.print(m.address()); Serial.print(" ");
        Serial.print(m.arg<int>(0)); Serial.println();

    });
    osc.subscribe("/need/reply", [](OscMessage& m)
    {
        Serial.println("/need/reply");

        int i = 12;
        float f = 34.56F;
        double d = 78.987;
        String s = "hello";
        bool b = true;

        osc.send(host, send_port, "/send", i, f, d, s, b);
    });

    // TODO: TBD
    // osc.publish(host, send_port, "/value", value);
    // osc.publish(host, send_port, "/millis", &millis);
}

void loop()
{
    osc.parse(); // should be called
}
```

devinroth, 2019-04-02:
I should add that I don't think it's a parsing problem. It doesn't crash when data is received.

devinroth, 2019-04-03:
I just tried out a different ESP32 board and it doesn't crash. There must be something wrong with the other board. Thanks for you help.

closed 2019-04-03 by devinroth