hideakitai/ArduinoOSC #8: esp32 + ethernet, example not working

himijendrix24, 2019-08-06:
Hi, not really an issue but it might be helpful for others.
I'm using an Olimex-ESP32-Gateway (esp32 + ethernet board) and the ethernet examples didn't work. The esp32 crashed right after sending osc because in the examples the class is initialized as OscSerial osc;.

This code works for me and you can live un-/replug ethernet on sender and receiver.

```
#include <ETH.h>
#include <ArduinoOSC.h>

bool eth_connected = false;
IPAddress local_IP(192, 168, 0, 50);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(192, 168, 0, 1);
IPAddress secondaryDNS(8, 8, 8, 8);

// ArduinoOSC (library examples don't work)
OscWiFi osc;
const char* host = "192.168.0.60";
const int recv_port = 8000;
const int send_port = 8010;

// Ethernet callbacks
void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      osc.begin(recv_port);
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void setup() {
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
}

void loop() {
  // ethernet connected
  if (eth_connected) {  
        // sending code
        // ....
  }

  else {
    // Connect ethernet
    Serial.println("attempting ethernet connection ...");
    ETH.begin();
    ETH.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS);
    delay(2000);
  }
}
```

vikbessems, 2019-08-14:
I have similar problems with sending osc messages on the huzzah esp 32.
Decoded Guru Meditation Error :
```PC: 0x400d29a4: ArduinoOSC::OscServerUdp ::parse() at D:\Documents\Arduino\libraries\ArduinoOSC/ArduinoOSC/OSCServer.h line 55
EXCVADDR: 0x00000000

Decoding stack results
0x400d29a4: ArduinoOSC::OscServerUdp ::parse() at D:\Documents\Arduino\libraries\ArduinoOSC/ArduinoOSC/OSCServer.h line 55
0x400d2b66: loop() at D:\Documents\Arduino\new_test/new_test.ino line 42
0x400d72a5: loopTask(void*) at C:\Users\vikbe\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3-rc1\cores\esp32\main.cpp line 19
0x40089241: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

This seems to point at: osc.parse() and  const size_t size = this->stream->parsePacket(); ```

hideakitai, 2020-05-19:
Sorry for late reply... is that an issue of example? I have updated this library and got much simpler, so please check it out. I will close this issue now, but if you still have problem, re-open and let me know. Thanks for your feedback.

closed 2020-05-19 by hideakitai