hideakitai/ArduinoOSC #9: OscWifi not working

lacouture100, 2019-11-09:
Hello @hideakitai,

What a wonderful library, however I can't seem to get it to work with the Arduino Nano 33 ioT or the mkr 1010 wifi. I was wondering if it is something I'm missing or if the library hasn't integrated these two. Anyway, I'll leave the error ahead.

Thank you! 

`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\lacou\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\lacou\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lacou\Documents\Arduino\libraries -fqbn=arduino:samd:nano_33_iot -vid-pid=0X2341_0X8057 -ide-version=10810 -build-path C:\Users\lacou\AppData\Local\Temp\arduino_build_613478 -warnings=none -build-cache C:\Users\lacou\AppData\Local\Temp\arduino_cache_511777 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.openocd.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -verbose C:\Users\lacou\Documents\Arduino\libraries\ArduinoOSC\examples\arduino\OscWiFi\OscWiFi.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\lacou\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\lacou\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lacou\Documents\Arduino\libraries -fqbn=arduino:samd:nano_33_iot -vid-pid=0X2341_0X8057 -ide-version=10810 -build-path C:\Users\lacou\AppData\Local\Temp\arduino_build_613478 -warnings=none -build-cache C:\Users\lacou\AppData\Local\Temp\arduino_cache_511777 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.openocd.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -verbose C:\Users\lacou\Documents\Arduino\libraries\ArduinoOSC\examples\arduino\OscWiFi\OscWiFi.ino
Using board 'nano_33_iot' from platform in folder: C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4
Using core 'arduino' from platform in folder: C:\Users\lacou\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4
Detecting libraries used...
"C:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino NANO 33 IoT\"" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\cores\\arduino" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\variants\\nano_33_iot" "C:\\Users\\lacou\\AppData\\Local\\Temp\\arduino_build_613478\\sketch\\OscWiFi.ino.cpp" -o nul
Alternatives for ArduinoOSC.h: [ArduinoOSC@0.2.2]
ResolveLibrary(ArduinoOSC.h)
  -> candidates: [ArduinoOSC@0.2.2]
"C:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino NANO 33 IoT\"" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\cores\\arduino" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.4\\variants\\nano_33_iot" "-IC:\\Users\\lacou\\Documents\\Arduino\\libraries\\ArduinoOSC" "C:\\Users\\lacou\\AppData\\Local\\Temp\\arduino_build_613478\\sketch\\OscWiFi.ino.cpp" -o nul
Error while detecting libraries included by C:\Users\lacou\AppData\Local\Temp\arduino_build_613478\sketch\OscWiFi.ino.cpp
Generating function prototypes...
"C:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino NANO 33 IoT\"" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\lacou\\AppData\\Local\\Arduino15\\packages\ [... cut ...]
c:\users\lacou\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:265:56: error: macro "max" passed 3 arguments, but takes just 2

     max(const _Tp& __a, const _Tp& __b, _Compare __comp)

                                                        ^

Multiple libraries were found for "ArduinoOSC.h"
 Used: C:\Users\lacou\Documents\Arduino\libraries\ArduinoOSC
Using library ArduinoOSC at version 0.2.2 in folder: C:\Users\lacou\Documents\Arduino\libraries\ArduinoOSC 
exit status 1
Error compiling for board Arduino NANO 33 IoT.`

hideakitai, 2020-05-19:
Sorry for late reply... Nano and MKR series uses `WiFiNINO` library right? I haven't support that because here in Japan, we can't use those series because of the "Radio Act". If you can test it in your own board, I can support them.

cansik, 2020-06-16:
As a simple workaround, just include the following define in your sketch:

```cpp
// quick fix to support Arduino Rev 2
#define OscWiFi ArduinoOSC::Manager<WiFiUDP>::getInstance()
```

If I find time later I'll include properly in the `ArduinoOSC.h`.

hideakitai, 2020-06-18:
Some boards have been supported thanks to @cansik . Thanks!

closed 2020-06-18 by hideakitai