hideakitai/ArduinoOSC #10: Esp32 not receiving OSC messages via demo python script

JamesNitsch, 2020-06-05:
Hello,

I've been trying to get your library to work for some time now, with no success. I'm running the default Arduino ino found in the examples alongside the python script also found in the examples.

I've verified that both devices are on the same network & band, and I've confirmed that the python script is sending the messages to the expected IP / port. The ESP32 device connects just fine, but does not receive any of the sent messages. I have not modified any of the ports, ips, or masks of the example code.

Any ideas as to why the messages may not be coming through? Thank you.

hideakitai, 2020-06-05:
Hi, in my environment, it works.
Can you check network connection again including subnet setting? Because ESP's IP is 192.168.1.201 but host is set as 192.168.2.20. If pinging and sending OSC are success (if ESP's serial port says that received OSCs), check firewall setting if you are trying that on Windows.

JamesNitsch, 2020-06-06:
Thank you for your response. I was able to resolve the issue with some help. Seems like I needed to use a 10.0.X.X subnet with my router. Everything is working well now!

closed 2020-06-06 by hideakitai