hideakitai/ArduinoOSC #11: Subscribeの実行がpublishのインターバルに何らかの影響を受けている？

hana, 2020-06-13:
こんにちは。最高に使いやすいライブラリの公開、ありがとうございます。

タイトルにもあります通り、Subscribeの応答速度がpublishの有無とインターバルによって変化してしまうような挙動をしております。こちらは使用上仕方のないことなのでしょうか？

具体的には、ESP32にて以下のようなコードを実行しているのですが、ESP32がOSCメッセージを受信した際に即時に応答せず、パケットが1秒程度溜まってから排出されるような挙動をしてしまっています。なおOSCメッセージはMaxから100msecごと`/foo X`の形で送っており、Xは100msecごとに0.01ずつインクリメントされるようになっています。
**コード**
```
void setup() {
  delay(1000);
  WiFi.begin(ssid, pwd);
  WiFi.config(ip, gateway, subnet);

  while(WiFi.status() != WL_CONNECTED) { Serial.print("."); delay(500); }
  Serial.begin(115200);

  OscWiFi.subscribe(port_a, "/foo",
        [&](const float& f)  {
            Serial.print("/foo ");
            Serial.println(f);
        }
    );

  OscWiFi.publish("224.0.0.1", port_b, "/active", 1.0) -> setIntervalMsec(1000.0f);
}

void loop() {
    OscWiFi.update(); 
}
```
[
[image]
](url) 

OscWiFi.publishの行をコメントアウトしてみたところ、以下のようになり、連続的な受信がままならない状況になりました。

[image]

publishとsubscribeに何らかの因果を感じたので、今度はpublishをアンコメントし、intervalを1 msecにしてみたところ、subscribeの応答速度が大変早まりました。

可能であればintervalを1000msのまま、subscribeはなるべく高速に使用したいのですが、そのようなことは可能なのでしょうか？お忙しいところお手数をおかけしますがご教授いただけると幸いです。
よろしくおねがいします。

hideakitai, 2020-06-14:
@hana ご報告ありがとうございます。こちらは想定外の挙動で、本来はきちんと動くはずなのですが、、、。すぐには確認できませんが、今日明日とかで確認してみたいと思います。publish subscribe 両方を登録した状態で、loop() 内で `OscWiFi.parse()` のみ (subscribeのみ実行される) と、`OscWiFi.post()` (publishのみ実行される)を、それぞれ実行した場合どうなるか試すことは可能でしょうか？`OscWiFi.update()` すると、parse() と post() を両方やっている状態です。

hana, 2020-06-14:
@hideakitai ご返答ありがとうございます。お教えいただいた方法など色々と試してみたところ、どうやら、私のWiFiルーターの問題だったようで、ルータを変更したら解決いたしました。お手数おかけいたしまして申し訳ありません。

加えて、一つ質問をよろしいでしょうか。最初（1000ms秒ごとにpublish, `loop()`内では`OscWiFi.update()`を使用しMaxで100msecごとにメッセージを送信）のプログラムを実行したところ、以下のようにいくつかのメッセージが300msec程度ごとに処理される時があるのですが、こちらも私の環境もしくはシリアル通信のボトルネックなどが原因である可能性が高いでしょうか？

[image]

お手数おかけしますがよろしくお願いいたします。

hana, 2020-06-14:
すみません、ネットワーク機器の問題かと思っていたのですが、元のネットワーク機器上で`OscWiFi.pubish()->setInterval(1.0)`の設定と共に使うと高速で動作するので、ネットワーク機器だけの問題ではないかもしれません。お手数ですがよろしくおねがいします。

hideakitai, 2020-06-15:
@hana いろいろ試していただいてありがとうございます。Arduino IDEのバージョンと、ESP32のバージョン教えてもらっても良いですか？手元だと、Arduino 1.8.12 + ESP 1.0.4 だとそもそも `subscribe` のみでも挙動がおかしく、そもそもUDPパケット自体がちゃんと飛んでこないです。 Arduino 1.8.12 + ESP 1.0.2 までバージョンを落とすと、手元では問題なく動作しているように見えます。試していただいても良いでしょうか？

hana, 2020-06-17:
@hideakitai ご連絡ありがとうございます。Arduino 1.8.12 + ESP 1.0.4のバージョンで使用しておりました。`Arduino 1.8.12 + ESP 1.0.2`で試したところ、無事動作いたしました。また昨日リリースされた`Arduino 1.8.13`に`ESP 1.0.4`の組み合わせでも同様に動いている様子です。

ご確認いただき本当にありがとうございます。

hideakitai, 2020-06-17:
@hana 無事動作しているようで良かったです。最新バージョンのご報告もありがとうございます。クローズして、概要を(つたない)英語でまとめておきますね。ありがとうございました！

hideakitai, 2020-06-17:
- Issue: sometimes `subscribe` is unstable (lost packet, delay timing, etc.)
- Reason: `Arduino IDE 1.18.12` + `ESP v1.0.4` couldn't receive raw udp packet correctly
- Soliution: use `ESP v1.0.2` for `Arduino IDE 1.18.12`
  - `Arduino IDE 1.18.13` + `ESP v1.0.4` seems to be working fine currently

closed 2020-06-17 by hideakitai