hideakitai/ArduinoOSC #12: fix bug for AVR boards in which memory layout is not correct.

cansik, 2020-06-16:
I am currently working with the Arduino Uno WiFI Revision 2 and experience weird bugs and behaviours. It really has problems to parse the incoming data which seems to be the `ceil4` related problem.

I guess it has to do with the following [todo line](https://github.com/hideakitai/ArduinoOSC/blob/452c80f633a517ef002ad3f90b94e87963f9b38f/ArduinoOSC/OscMessage.h#L218):

```
// TODO: fix bug for AVR boards in which memory layout is not correct.
```

Is it possible to create a workaround for this (even we have some performance issues)?

cansik, 2020-06-16:
Wow...it seems that it depends on how the source code is structured. Adding some Serial prints to the decoder, and everything works fine:

```
Address: /hello/world
Address Begin:16152
Address End:16164
TypeTags Begin: 16168
TypeTags End: 16171
TypTags Length: 12
IsZeroPadding Correct: 1
First Char: ,
```

Removing them breaks the memory model.

Now I understand this line of comment:

```
// if bool dummy; is added in this class, building from raw data fails because of ceil4 return not correct pointer address
```

cansik, 2020-06-16:
Ok, here are the bugs (actually two):

```
Address: /hello/world
Address Begin:16151
Address End:16163
Address Length:12
TypeTags Begin: 16164
TypeTags End: 16164
TypTags Max Length: 15
IsZeroPadding Correct: 0
RAW: 2F(/) 68(h) 65(e) 6C(l) 6C(l) 6F(o) 2F(/) 77(w) 6F(o) 72(r) 6C(l) 64(d) 0( ) 0( ) 0( ) 0( ) 2C(,) 69(i) 69(i) 0( ) 
0. Char:  
1. Char:  
2. Char:  
3. Char: ,
4. Char: i
```

It seems that some OSC implementations send 4 bytes padding, even the address is already a multiple of 4. I have tested this with TouchOSC and oscP5. This is the bug in the first example.

Most of the libraries I have worked with look for the type delimiter `,` which would be fine here too with a `memchr`.

And the second bug:

```
RAW: 2F(/) 68(h) 65(e) 6C(l) 6C(l) 6F(o) 2F(/) 77(w) 6F(o) 72(r) 6C(l) 64(d) 0( ) 0( ) 0( ) 0( ) 2C(,) 69(i) 69(i) 0( ) 
Address Begin:16153
Address End:16165
Address Length:12
Address Begin:/
Address: /hello/world
TypeTags Begin: 16168
TypeTags End: 16168
TypTags Max Length: 13
IsZeroPadding Correct: 0
0. Char:  
1. Char: ,
2. Char: i
3. Char: i
4. Char:  
```

Here the problem is that you are doing wrong math with your pointers. You are rounding up the pointers to a multiple of 4, but not by its number in the memory. 

```
ceil4(16165) = 16168
```

But you have to ceil4 the bits relative to it's starting: 

```
ceil4(16165 - 16153) + 16153 = 16165
```

And again then we would have 4 bytes problem as described above. Also the `memchr` would fix this problem.

hideakitai, 2020-06-17:
Thanks for your PR! @cansik

closed 2020-06-17 by hideakitai