hideakitai/ArduinoOSC #15: Possible deadlock in subscribe or send

cansik, 2020-06-17:
While flooding the Arduino UNO WiFi Rev2 with a lot of osc messages, it sometimes happens that the main loop just freezes. It's not an Arduino crash but really a blocking and my wild guess is the `OscWiFi.update()`.

I am also sending back data, so it could also be a problem with both connected:

```cpp
OscWiFi.subscribe(IN_PORT, "/tes/read, [](const OscMessage & m) {
  OscWiFi.send(m.remoteIP(), OUT_PORT, "/test/answer", m.arg<int>(0), 42);
});
```

Have you already experienced something similar?

hideakitai, 2020-06-18:
Mmm.. it can be occur if:

1. Too many packets on subscribe and publish (send)
2. ‘WiFiUDP::parsePacket’ cannot receive correctly
  - WiFi condition
  - Board bugs (encountered in #11)
3. Library bugs??

So can you let me know about
- versions you use
- how many and how often send/receive packet
- check if ‘parsePacket’ returns byte size correctly
to reproduce the problem.

hideakitai, 2020-06-18:
‘parsePacket’ is called in OscSever.h ‘Server::parse()’.

cansik, 2020-06-18:
Ok, I guess it has to do with the following bug: https://github.com/arduino-libraries/WiFiNINA/issues/108
It seems that sometimes WiFiNINA `upd.parsePacket` is returning the wrong package size (sometimes 1500 instead of 24). I will write my findings there.

Sorry for bothering you!

closed 2020-06-18 by cansik