hideakitai/ArduinoOSC #16: Buffer gets overwritten while decoding

cansik, 2020-06-18:
Is it possible that a buffer can get overwritten while still parsed?

Sending following messages (quite fast behind each other).
The number indicates the length in bytes:
```
/iad/led/color iii 36
/iad/imu/rpy i 24
```

On the arduino, when falling into [this error-check](https://github.com/hideakitai/ArduinoOSC/blob/master/ArduinoOSC/OscMessage.h#L253) (regarding too few arguments), I print out the current buffer (`storage.end() - storage.begin()`), first the byte value and then the corresponding charachter:

```
Size: 36
47|/ 105|i 97|a 100|d 47|/ 105|i 109|m 117|u 47|/ 114|r 112|p 121|y 0|  0|  0|  0|  44|, 105|i 0|  0|  0|  0|  1| 101|e 0|  0|  0|  49|1 0|  0|  0|  0|  0|  0|  0|  -1|⸮
```

Well, that looks like the 36 byte buffer of the `/iad/led/color` packet, but filled with the values of the `/iad/imu/rpy` packet.

I am super confused now...either I am not reading the right data for debug or there is some concurrent work going on. Sending the same package behind each other of course makes no difference, because we are overwriting the same buffer.

After one of these messages, the `stream.parsePacket()` returns always `65524`.

Here an example sketch, which let me recreate this problem:

```c
#include <SPI.h>
#include <WiFiNINA.h>
#include <WiFiUdp.h>
#include <utility/wifi_drv.h>

#include <ArduinoOSC.h>

#include "secrets.h"

#define OscWiFi ArduinoOSC::Manager<WiFiUDP>::getInstance()

#define IN_PORT 8000

int keyIndex = 0;
int status = WL_IDLE_STATUS;

void setup() {
  Serial.begin(9600);

  // connect wifi
  while (status != WL_CONNECTED) {
    Serial.print("Connecting to SSID: ");
    Serial.println(ssid);
    status = WiFi.begin(ssid, keyIndex, pass);
  }

  // setup inputs
  OscWiFi.subscribe(IN_PORT, "/test/rgb", [&](int& r, int& g, int& b) {
    // do nothing
  });

  OscWiFi.subscribe(IN_PORT, "/test/w", [&](int& w) {
    // do nothing
  });

  Serial.println("everything setup to test!");
}

void loop() {
  OscWiFi.update();
}
```

Just send (with any OSC client) `/test/rgb` and `/test/w` alternately in a high framerate.

cansik, 2020-06-18:
Ok, the same seems to happen in [CNMAT/OSC](https://github.com/CNMAT/OSC) as well, so it is really a WiFiNINA problem. But would be great if you could test it on another board which if you have on available.

hideakitai, 2020-06-19:
Ok, on my ESP32, it works fine on 60fps for each.

It's not completely relate to this issue, the packet storage is based on the ring buffer with fixed size if you use WiFiNINA boards (because they cannot use STL). Currently it is [set to 64 bytes by default](https://github.com/hideakitai/ArduinoOSC/blob/f3920d96b1dc15770785edafd100c559d3b33510/ArduinoOSC/OscTypes.h#L20), so if you want to use bigger message than it, please use change this macro.

However, all osc message has 64 bytes buffer storage in each, it's not related to this issue... just FYI. Please let me know if you have the progress!

cansik, 2020-07-09:
Has been fixed in https://github.com/arduino/nina-fw/pull/49! 🎉

closed 2020-07-09 by cansik

hideakitai, 2020-07-09:
Great! Thanks for the report!