hideakitai/ArduinoOSC #19: ArduinoOSC requires Ethernet library even when only WiFi is used

superkartoffel, 2021-05-14:
Stumbled upon this when I tried your library for the first time. Not sure if you consider this a bug, but since Ethernet is not stated as a required dependency, I do :)

steps to reproduce in Arduino IDE:
* make sure you have Ethernet lib NOT installed
* load arduino/OscWiFi example project
* select any board that supports Ethernet (e.g. ESP32)
* try to compile

Expected result:
It should compile without errors.

Actual result:
```
In file included from /home/kartoffel/Arduino/libraries/ArduinoOSC/examples/arduino/OscWiFi/OscWiFi.ino:2:0:
/home/kartoffel/Arduino/libraries/ArduinoOSC/ArduinoOSC.h:54:22: fatal error: Ethernet.h: No such file or directory
compilation terminated.
```

superkartoffel, 2021-05-14:
A possible solution could be to split the header up into
ArduinoOSC_WiFi.h
ArduinoOSC_Ethernet.h

which set some defines and then include
ArduinoOSC.h

hana, 2021-05-15:
Just sharing, this also happened in my environment, where PlatformIO, esp-idf, and Arduino (as a component) is used. 
So I tried installing the latest(v2.0.0) [Ethernet Library](https://github.com/arduino-libraries/Ethernet), ending up with a compile error on it.

hideakitai, 2021-05-24:
@superkartoffel @hana Thank you for reporting, fixed in v0.3.22!

closed 2021-05-24 by hideakitai