hideakitai/ArduinoOSC #25: Is it possible to have a non-fixed IP ?

stephanschulz, 2021-09-14:
Thanks fro making this library. It works great.

Is there an option to get an IP assigned via DHCP? Or do I always need to hard code one?

Thanks for any advice.

hideakitai, 2021-09-20:
@stephanschulz Hi, sorry for the late reply. 

You should specify IP or hostname. It is because Arduino's `Client` class only supports them. So please try to set the hostname or get the assigned IP from your device.

stephanschulz, 2021-09-21:
I believe i have solved this by

```
WiFi.begin(ssid, pwd);
WiFi.config(0U, 0U, 0U);
```
which i found in this thread: https://stackoverflow.com/questions/40069654/how-to-clear-static-ip-configuration-and-start-dhcp

hideakitai, 2021-09-21:
Ah, is your device a receiver? IP is not required to subscribe OSC. But sender should know the host name or IP of your device. 

Closing this issue but please feel free to reopen if you still have any problem.

closed 2021-09-21 by hideakitai