hideakitai/ArduinoOSC #26: Unable to parse or post any OSC message on WiFi

Markdood88, 2021-11-26:
I'm using an Arduino MKR Wifi 1010 module, and the example with the WiFi.h library. I'm able to connect to the network, set an ip address, and make publish messages (although they are only 2 messages per second, not the 30hz mentioned). However when I send OSC back to the Arduino IP, on any of the reports, it doesn't print out any data to the serial monitor.

I'm wondering how to parse and post custom messages to 3rd party applications like TouchDesigner.

hideakitai, 2021-11-26:
Hi, do you use Windows? Have you tried to disable the firewall?

Markdood88, 2021-11-26:
I'm on a mac. 

Side note: I'm not sure if the received OSC messages need to have a specific format to them to be received?

hideakitai, 2022-01-02:
@Markdood88 Sorry for the late reply. I've tried to use MKR WiFi 1010, and this issue seems to be the problem of WiFiNINA (WiFi library for MKR WiFi1010). This library works fine with ESP32.

With MKR WiFi 1010, publishing OSC messages from MKR -> PC works fine. But sending OSC PC -> MKR is very unstable. The IP address setting causes this issue in my quick survey (I don't know why), and ping is unsuccessful in some IPs.

In my case, 192.168.0.42 works fine, but 192.168.0.99, 192.168.0.100, 192.168.0.101, and 192.168.0.201 does not. Could you try several IPs?

In addition, you can work with TouchDesginer, of course (I also use TouchDesigner to send/receive OSC with Arduino)

hideakitai, 2022-03-07:
I am closing this issue because no further information has been provided. Maybe this is not a problem of library, but a problem of WiFiNINA

closed 2022-03-07 by hideakitai