hideakitai/ArduinoOSC #28: run ArduinoOSCEther with OLIMEX ESP32 POE

ChristopheLebreton, 2022-05-06:
Hi, I try to use OLIMEX ESP32 POE with ArduinoOSCEther.
With this device ETH_LAN8720 example works through ETH.h provide by library Espressif ESP32 but it doesn't work with Ehternet.h used with ArduinoOSCEther.

There's a way to run ArduinoOSC through Ethernet with kind of this board ?

best ;)

ChristopheLebreton, 2022-05-06:
Library below also works and like used UDP perhaps there's a way to use with your's ? :

AsyncUDP_WT32_ETH01 is a Async UDP library for the WT32_ETH01 (ESP32 + LAN8720)
Based on and modified from ESPAsyncUDP Library (https://github.com/me-no-dev/ESPAsyncUDP)
Built by Khoi Hoang https://github.com/khoih-prog/AsyncUDP_WT32_ETH01

hideakitai, 2022-05-06:
Of course I can, but I don't have enough time and don't have a plan to support like it.

rsmck, 2023-03-22:
Hi @ChristopheLebreton did you find a solution to this? I've got the same issue.

I didn't realise the OLIMEX ESP32-POE used ETH.h not Ethernet.h.

rsmck, 2023-03-22:
In case anyone else stumbles upon this thread... `ETH.h` is derived from the usual WiFi libraries. 

I don't have the C skills to do this properly and submit a pull request, but an extremely hacky solution (that works) is as follows (maybe this will provide enough info for someone more competent to make official support.

In `ArduinoOSCWiFi.h` change line 11 from `#include <WiFi.h>` to `#include <ETH.h>`

In `ArduinoOSCCommon.h` comment out the if {} statement around line 60 that checks `if (WiFi.getMode() != WIFI_OFF)`, mine now looks simply like this;

```
        void subscribe(const uint16_t port, const String& addr, Ts&&... ts) {
            OscServerManager<S>::getInstance().getServer(port).subscribe(addr, std::forward<Ts>(ts)...);
        }
```

This allows ArduinoOSCWiFi (yes, the WiFi version, not the Ether one) to work as expected over Ethernet on the OLIMEX ESP32 POE. 

It does, of course, mean you can't use the WiFi interface... but for my application that didn't matter so I haven't investigated it any further. 

Hopefully this helps someone else.

hideakitai, 2023-08-07:
I will support ETH soon.

ChristopheLebreton, 2023-08-11:
Hi, 

very good news let me know if you want I test it ;)

Best

Chris

hideakitai, 2023-08-13:
`ETH.h` support added. Please see `examples/arduino/OscETH/OscETH.ino` for how to use it.

closed 2023-08-13 by hideakitai