hideakitai/ArduinoOSC #47: RP2040 not sending wifi when in station mode

MrFrangipane, 2023-10-25:
Hi, 

Thank you for this very nice library.

I wanted to connect to an OSC server (desktop computer with Python OSC server) through Wifi with a Raspberry Pi Pico (being the client).

I copied some parts of your example https://github.com/hideakitai/ArduinoOSC/blob/main/examples/arduino/OscWiFi/OscWiFi.ino but kept getting "wifi is not connected" messages in the log.

I modified lines `71`, `90`, and `102` of `ArduinoOSCCommon.h` as follows

````
if ((WiFi.status() == WL_CONNECTED) || (WiFi.getMode() == WIFI_STA)) {
````

instead of 

````
if ((WiFi.status() == WL_CONNECTED) || (WiFi.getMode() != WIFI_STA)) {
````

and now it works.

I'm not really sure why mode should not be `WIFI_STA` at all times, especially when line `56` of the example sets it to `WIFI_STA` ?

Thank you for your help,
All the best

hideakitai, 2023-11-14:
Maybe your library version is old. The latest release is fixed as you suggested. 

https://github.com/hideakitai/ArduinoOSC/blob/97082882a0cb8e774ba152a6cc5087704cdc244e/ArduinoOSC/ArduinoOSCCommon.h#L71

`WiFi.status() == WL_CONNECTED` is for Station Mode. It checks if the board is connected to WiFi AP. `WiFi.getMode() != WIFI_STA` is for AP Mode.

hideakitai, 2023-11-14:
Ah, sorry you modified from `WiFi.getMode() != WIFI_STA` to `WiFi.getMode() == WIFI_STA` ? I think your board is not correctly connected to WiFi AP (`WiFi.status() == WL_CONNECTED` is false) or the bug of WiFi library of Pi Pico

closed 2023-11-14 by hideakitai

MrFrangipane, 2023-11-14:
Hi, 

Thank you for your response :)

I'm not sure to understand: the mode should not be station after connection is successful ?

Anyways, I have switched the project to C/C++ SDK and use this library https://github.com/madskjeldgaard/PicoOSC

All the best !

hideakitai, 2023-11-15:
> I'm not sure to understand: the mode should not be station after connection is successful ?

No. The meaning of this if statement is:

- `WiFi.status() == WL_CONNECTED` is for Station Mode. It checks if the board is connected to the WiFi AP
- `WiFi.getMode() != WIFI_STA` is for AP Mode. Does not care about if the board is connected to WiFi AP or not (because ESP32 itself is AP)

So, if you use station mode, it checks if the board is connected to an existing AP. But it doesn't care if you use AP mode.

hideakitai, 2023-11-15:
I'll confirm with Pi Pico W later :)

MrFrangipane, 2023-11-17:
Hi,

Thank you very much !