hideakitai/ArduinoOSC #52: In  OscWiFiAP example, no ip obtained.

bmentink, 2024-03-05:
Hi,
I tried this example from .pio PlatformIO on ESP32.

The example sets up an AP just fine, and I can connect with the defaults password.
However, it just sits these trying to get an IP and eventually times out .. I tested with a Motorola Android Phone.

Any idea's to fix this appreciated. Cheers.

hideakitai, 2024-03-05:
How about using Arduino IDE？

bmentink, 2024-03-05:
I still hate Arduino IDE with a vengeance :)  But will try ..

bmentink, 2024-03-05:
Found it. Installed the library in Arduino IDE.
However tried it, and  that is worse, won't even except the default password, so can't get as far ..
Used the password in the code "0123456789"

hideakitai, 2024-03-05:
Sorry, I couldn't understand… You can change the passcode for your own in your code.

bmentink, 2024-03-05:
Yes I know that, I just used the default..

hideakitai, 2024-03-05:
I couldn't understand what the problem was. How do you try to get IP? Please explain with your complete code which can be compilable and reproduce the problem.

bmentink, 2024-03-05:
As I said, my phone connects to the AP, it tries to get IP from AP but
cannot.. What don't you understand about that?

A client must be able to get a client from an AP using DHCP.. You know this
right?

hideakitai, 2024-03-05:
I couldn't understand “who” wants to get IP. You can see IP in your console with fixed IP. It’s not a library scope, but you can use DHCP without a fixed IP. You can also add a hostname to your AP. Please refer to APIs for more details.

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiAP.h

bmentink, 2024-03-05:
Forget it

hideakitai, 2024-03-05:
If you have solved the problem, please close this issue.

closed 2024-03-05 by bmentink

hideakitai, 2024-03-05:
Understood. Your problem is not with this library but with the usage of the WiFi library itself. I'll post the ChatGPT's response at the end.

-

To use an ESP32 in WiFi Access Point (AP) mode and automatically assign IP addresses to connected clients via DHCP, follow the instructions below. Here’s an example of a working Arduino code for this purpose. By using this code, the ESP32 will function as a WiFi access point and will automatically allocate IP addresses to devices that connect to it.

```
#include <WiFi.h>

const char *ssid = "ESP32-Access-Point"; // SSID of the access point
const char *password = "123456789"; // Password of the access point, minimum 8 characters

void setup() {
  Serial.begin(115200);
  
  // Start WiFi in access point mode
  WiFi.softAP(ssid, password);
  Serial.println("Access Point Started");
  Serial.print("IP Address: ");
  Serial.println(WiFi.softAPIP()); // Outputs the IP address of the ESP32
}

void loop() {
  // Here, you can add code to monitor connections, or to perform other tasks.
}
```

This code sets up the ESP32 as an access point with the SSID ESP32-Access-Point and the password 123456789. The IP address of the ESP32 is obtained and displayed in the Serial Monitor using the WiFi.softAPIP() function.

The ESP32’s DHCP server automatically assigns IP addresses to connected clients. This feature is enabled by default and requires no additional configuration.

bmentink, 2024-03-05:
` By using this code, the ESP32 will function as a WiFi access point and will automatically allocate IP addresses to devices that connect to it.`

But it doesn't allocate the IP THAT IS MY POINT. The phone that tries to connect CAN"T GET AN IP !!!!

hideakitai, 2024-03-05:
It’s a problem with the usage of the WiFi library or an issue with the WiFi library itself. Because it’s not a scope of this library, I recommend asking it to the esp32-arduino repo.