hideakitai/ArduinoOSC #55: Unable to send OSC from Arduino Uno R4 WiFi

marcvillanuevamir, 2024-08-02:
Hi!
I am running the OscWifi example on my Arduino Uno R4 WiFi, and using your example maxpatch to communicate with it from my Mac. The issue is that the Arduino receives every message I send to it, but I can't get any message from it. For instance, when sending /need/reply, the message is printed on the Arduino IDE monitor, but I get nothing in Max. Am I missing something? Thank you!!

hideakitai, 2024-08-02:
What OS do you use? Windows? Have you configured your firewall and such incoming packet restriction settings?

marcvillanuevamir, 2024-08-02:
I use iOS, running Sonoma 14.6. My network's firewall is off. I don't know if there is anything else to configure in terms of clearing restrictions.

hideakitai, 2024-08-02:
How about other network configurations like IP, subnet, gateway, or router? Sometimes, your router may block some ports/packets. You can check if the OSC packet comes to your PC using Wireshark.

marcvillanuevamir, 2024-08-02:
I'm not so familiar with networks in-depth, but I double-checked that the subnet and gateway configurations of my computer match those in the Arduino code. I ran Wireshark while sending repeated messages of /need/reply and I couldn't see anything coming from the Arduino's IP. I see, though, periodic ARP messages coming from the Arduino (I checked its MAC address), as can be seen in this screenshot:

[image]

marcvillanuevamir, 2024-08-02:
I just figured it out! I didn't change the "host" on the arduino code. I wrote down the IP of my computer and now I got full communication! Those ARP messages that Arduino kept sending looking for a host gave me the clue :) Thank you so much for your help!

closed 2024-08-03 by hideakitai