hideakitai/ArduinoOSC #56: Timestamp support?

jhurliman, 2024-08-07:
I see the message in the README about timestamps being ignored. I’m looking to add time synchronization and receive scheduled OSC messages on a dozen ESP32s. I’m happy to submit a PR to add timestamp support, but wanted to check:

1. Is this something you’d be interested in merging?
2. Where should I look in the codebase to add timestamp support?

hideakitai, 2024-08-14:
Hi, @jhurliman 

Thank you very much for considering your contribution. Yes, I'm interested in it. For the current status, 

- Decoding timestamp in the bundle probably works (refer [here](https://github.com/hideakitai/ArduinoOSC/blob/b5b5daa7fbfd4830d447da53321cb92c57d4e021/ArduinoOSC/OscDecoder.h#L55-L84))
- The timestamp using when encoding bundle without your own `TimeTag` is dummy value (refer [here](https://github.com/hideakitai/ArduinoOSC/blob/b5b5daa7fbfd4830d447da53321cb92c57d4e021/ArduinoOSC/OscEncoder.h#L48-L57))
  - You can pass your own `TimeTag` when making the bundle (`begin_bundle(your_timetag)`)

...Huh? You could use it if you wanted to... maybe!

hideakitai, 2024-08-14:
If you can confirm you can use `TimeTag`, we should fix the comment in the README

jhurliman, 2024-08-14:
I don't see an example of how to send a manually constructed `OscMessage`, could you point me in the right direction?

```
OscMessage msg_out;
OscEncoder wr;

// Get the current time (ESP32)
const uint64_t now = esp_timer_get_time();

wr.init().begin_bundle(TimeTag(now));
wr.encode(msg_out.init(OSC_TIME_RESPONSE).push(now));
wr.end_bundle();
// no matching function for call to 'arduino::osc::Manager<WiFiUDP>::send(const String&, const uint16_t&, OscMessage&)'
OscWiFi.send(msg_in.remoteIP(), OSC_SEND_PORT, msg_out);
```

hideakitai, 2024-08-16:
Supported (but not tested) sending with timetag in main branch https://github.com/hideakitai/ArduinoOSC/pull/57

Please try

```c++
    const uint64_t now = esp_timer_get_time();
    const auto tt = OscTimeTag(now);
    OscWiFi.send(host, send_port, tt, "/msg", 1, 2.2f, "hello");
```

hideakitai, 2024-08-16:
Ah sorry, it was not a bundle

hideakitai, 2024-08-16:
The bundle is now supported in the main branch. Can you try this example? @jhurliman 

https://github.com/hideakitai/ArduinoOSC/blob/main/examples/arduino/OscWiFiBundle/OscWiFiBundle.ino

hideakitai, 2024-08-25:
If you still need help, please reopen this issue.

closed 2024-08-25 by hideakitai